openbci-archive/Docs #15: Documentation does not mention accelerometers much

yannrenard, 2016-04-29:
The documentation does not mention much about how accelereometers can / should be used. Interesting information to find in the documentation includes :
 - sampling rate (which is supposedly different than that of EEG)
 - scale factor to apply to the digital values (which is also supposedly different than that of EEG)
 - how to deal with accelerometers when daisy is used

retiutut, 2019-05-20:
Looking to finally add documentation to fix this issue when Accelerometer is added to the Networking Widget. There is a new Doc for Widget Tutorials, which is a great place for information on the Accelerometer Widget.

retiutut, 2019-06-26:
https://github.com/OpenBCI/Docs/blob/master/OpenBCI%20Software/02_GUI_Widget_Guide.md#accelerometer

closed 2019-06-26 by retiutut