openbci-archive/Docs #76: Wrong Link for Linux Radio Programming

baffo32, 2017-04-06:
Not sure how to reopen an issue, so just a note that the fix for #73 was incorrect.  What is actually needed is a script that wraps the windows binary with wine, not the windows binary itself.

closed 2019-04-20 by retiutut

baffo32, 2019-04-20:
@retiutut this issue appears to not have been addressed.  Could you comment on it being closed?

retiutut, 2019-04-20:
It sounds like you know how to fix it and it has almost been two years. You are welcome to edit the Docs and submit a pull request.

baffo32, 2019-04-20:
It sounds like you guys have a lot on your hands.  I’m on a mobile phone, here.  Ideally the Linux doc instructions should be tested on Linux with the relevant chips connected.  The instructions right now do not work because they reference the wrong file.  Would you be willing to reopen #73 so that somebody other than me can learn how to make them work?  Right now I may be the only person who knows of this issue, and I’m handling too many other things to be certain I’ll remember it when I’m able to address it.  (The fix is also pretty easy if you look at the end of #73)

retiutut, 2019-04-20:
I still don't understand what exactly the issue is, but if you already know, just type it out in the docs instead of re-explaining the problem in comments.

baffo32, 2019-04-20:
I am on a broken, slow mobile phone in a rural area without my computer and struggling with serious psychiatric issues, retiut.  Loading an interactive web editor causes my phone to thrash to almost a halt due to low ram.  Is there a reason you are closing unresolved issues?

retiutut, 2019-04-20:
Old issues. The community is built on contributions from its members.

retiutut, 2019-04-20:
Demanding others do the work when you seem to know a solution seems kind of weird, especially when its been so long.

retiutut, 2019-04-20:
Or provide more relevant details. All of this back and forth about what works or does not work is unclear.

baffo32, 2019-04-20:
I added #196 .  Please understand that this issue is the last available information on how to build the cyton radio software on Linux successfully.

The problem was that the OpenBCI servers deleted a supporting file that the instructions referenced.  When I noted this, I accidentally recommended the wrong file to replace it, and they have been broken ever since.  The correct file appears to be something like https://github.com/OpenBCI/Docs/issues/73#issuecomment-288889646 , but surely there is somebody at openbci who was around when that original file was put up or lost and could restore it?

I think of issues as places for people to collaborate on information.  I guess you’re right that it’s been a long time =S

retiutut, 2019-04-20:
Now you are a contributor!!!!

retiutut, 2019-04-20:
I was still able to download an RFLoader.exe.

baffo32, 2019-04-20:
To clarify with more details - the radio uploaded is only working for windows, so it needs to be run under wine in Linux.  There used to be a wrapper script to automate this, but it has disappeared and the documentation still references it.

I was trying to build the radio software and inferred it should be the .exe to download, two years ago, but I posted the issue prematurely and realized once I got it working better that it was not the windows binary that was lost but rather the Linux way of launching it.  I posted the updates to these issues but no reply for two years.  I have serious psychological issues around this project and it is very hard for me to contribute.

retiutut, 2019-04-20:
Looks like it should work! Easy!

retiutut, 2019-04-20:
If it doesn't someone will make a new issue and we will fix it!