openbci-archive/Docs #119: Mark IV closeup images of Cyton pins, wrong SRB pin

wjcroft, 2018-01-31:
On this thread in the forum:

http://openbci.com/forum/index.php?p=/discussion/comment/7536#Comment_7536

jeyem makes this comment:

> jeyem 3:07AM 141.20.217.36
> @wjcroft: What 1927student is referring to are these two pictures in the assembly instructions. This is obviously misleading for many newbees.
> 
> https://raw.githubusercontent.com/OpenBCI/Docs/master/assets/images/MarkIV/MarkIV_cyton_connects_nocover.JPG
> https://raw.githubusercontent.com/OpenBCI/Docs/master/assets/images/MarkIV/MarkIV_cyton_connects_cover.JPG

The SRB connection should show pin CLOSEST to board (SRB2), not SRB1. Bias can use either pin.

Files are:

https://github.com/OpenBCI/Docs/blob/master/assets/images/MarkIV/MarkIV_cyton_connects_cover.JPG
https://github.com/OpenBCI/Docs/blob/master/assets/images/MarkIV/MarkIV_cyton_connects_nocover.JPG

aj-ptw, 2018-01-31:
Oh my!!!!!! @conorrussomanno

RebeccaMurray, 2018-02-09:
Just added images of the correct configuration (using the bottom SRB2 pin). Sorry about that!

closed 2018-02-09 by RebeccaMurray