openbci-archive/Docs #120: Errors when Running GUI in Processing

rcsofficial, 2018-02-01:
I am getting these errors when running Processing though it still proceeds to open the GUI. I am using Acer Aspire E15 with Intel i7 5500U, 2.4GHz, Windows 10 64-bit

`
The sketch has been automatically resized to fit the screen resolution
Welcome to the Processing-based OpenBCI GUI!
Last update: 9/5/2016
For more information about how to work with this code base, please visit: http://docs.openbci.com/OpenBCI%20Software/
Graphics & GUI Library: ControlP5 2.2.6 infos, comments, questions at http://www.sojamo.de/libraries/controlP5
attempting to load images for presentation...
DONE loading images!
OpenBCI_GUI::Setup: Is RX mulitcast: false
OpenBCI_GUI::Setup: Has RX joined multicast: false
RunnableTask.run(): A caught exception occured on thread main-Display-.windows_nil-1-EDT-1: RunnableTask[enqueued true[executed false, flushed false], tTotal 0 ms, tExec 0 ms, tQueue 0 ms, attachment null, throwable java.lang.RuntimeException: Waited 5000ms for: <75db1984, 2cdaac89>[count 2, qsz 0, owner <main-FPSAWTAnimator#00-Timer0-FPSAWTAnimator#00-Timer1>] - <main-Display-.windows_nil-1-EDT-1>]
java.lang.RuntimeException: Waited 5000ms for: <75db1984, 2cdaac89>[count 2, qsz 0, owner <main-FPSAWTAnimator#00-Timer0-FPSAWTAnimator#00-Timer1>] - <main-Display-.windows_nil-1-EDT-1>
	at jogamp.common.util.locks.RecursiveLockImpl01Unfairish.lock(RecursiveLockImpl01Unfairish.java:198)
	at jogamp.newt.WindowImpl$ResizableAction.run(WindowImpl.java:2118)
	at jogamp.newt.DisplayImpl.runOnEDTIfAvail(DisplayImpl.java:450)
	at jogamp.newt.WindowImpl.runOnEDTIfAvail(WindowImpl.java:2782)
	at jogamp.newt.WindowImpl.setResizable(WindowImpl.java:2154)
	at com.jogamp.newt.opengl.GLWindow.setResizable(GLWindow.java:370)
	at processing.opengl.PSurfaceJOGL$5.run(PSurfaceJOGL.java:452)
	at com.jogamp.common.util.RunnableTask.run(RunnableTask.java:127)
	at jogamp.newt.DefaultEDTUtil$NEDT.run(DefaultEDTUtil.java:375)
DefaultEDT.run(): Caught exception occured on thread main-Display-.windows_nil-1-EDT-1: RunnableTask[enqueued false[executed true, flushed false], tTotal 5002 ms, tExec 5002 ms, tQueue 0 ms, attachment null, throwable java.lang.RuntimeException: Waited 5000ms for: <75db1984, 2cdaac89>[count 2, qsz 0, owner <main-FPSAWTAnimator#00-Timer0-FPSAWTAnimator#00-Timer1>] - <main-Display-.windows_nil-1-EDT-1>]
java.lang.RuntimeException: Waited 5000ms for: <75db1984, 2cdaac89>[count 2, qsz 0, owner <main-FPSAWTAnimator#00-Timer0-FPSAWTAnimator#00-Timer1>] - <main-Display-.windows_nil-1-EDT-1>
	at jogamp.common.util.locks.RecursiveLockImpl01Unfairish.lock(RecursiveLockImpl01Unfairish.java:198)
	at jogamp.newt.WindowImpl$ResizableAction.run(WindowImpl.java:2118)
	at jogamp.newt.DisplayImpl.runOnEDTIfAvail(DisplayImpl.java:450)
	at jogamp.newt.WindowImpl.runOnEDTIfAvail(WindowImpl.java:2782)
	at jogamp.newt.WindowImpl.setResizable(WindowImpl.java:2154)
	at com.jogamp.newt.opengl.GLWindow.setResizable(GLWindow.java:370)
	at processing.opengl.PSurfaceJOGL$5.run(PSurfaceJOGL.java:452)
	at com.jogamp.common.util.RunnableTask.run(RunnableTask.java:127)
	at jogamp.newt.DefaultEDTUtil$NEDT.run(DefaultEDTUtil.java:375)
Instantiating hub object...
Hub: clientEvent: handshake complete`

aj-ptw, 2018-02-28:
Are you using an internet proxy?

kallander, 2018-04-19:
Did you solve this? I'm getting a similar message..

aj-ptw, 2018-04-19:
I think this is when your system can’t handle the graphics library, can you describe your computer?

aj-ptw, 2018-04-19:
You added code and it stopped working?

kallander, 2018-04-19:
I added a library and some code, that I know works, https://github.com/robosolt/Processing-Sphero-Lib 
When just trying to get the program running I only added some lines to connect the sphero (and global variables etc) but what I get is a message similar to the one above. I've managed to specify the problem to something with the communication in the ports. Maybe there is a double in some of the variables...  I'm kinda new to the programming part..

retiutut, 2019-04-20:
Should be resolved by recent fixes in the GUI targeting startup behavior.

closed 2019-04-20 by retiutut