openbci-archive/Docs #134: Can't run GUI in Processing

BCIstorm, 2018-03-31:
Using windows 10 Pro 64bit, Processing 3.3.7 and GUI v3.2.0

Trying to put LSL to work, and after reading that there's some problems in the Standalone application I tried running the GUI from processing.

But, although OpenBCI Hub is running, the GUI says:
"Please launch OpenBCI Hub prior to launching this application."
And I can't not see the menu to choose the filename, COM port, etc clicking on LIVE( from Cyton) menu line.

Any clues?

Skalbeard, 2018-04-18:
Same! Can`t figure what to do. Neither standalone download, nor the installed version of OpenBCI hub seems to be picking up the device.

TimoteoMendes, 2018-06-01:
Me neither!
Have follow all the tutorial more than once update my processing, driver everything but no luck!

the hub give all this errors:
[image]

and the processing this:
No library found for ddf.minim
No library found for ddf.minim.ugens
No library found for grafica
No library found for netP5
No library found for oscP5
No library found for hypermedia.net
No library found for gifAnimation
No library found for controlP5
No library found for ddf.minim.analysis
Libraries must be installed in a folder named 'libraries' inside the sketchbook folder (see the Preferences window).

Can someone from OpenBCI explain way?
Thanks.
Tim

xXBannedXx, 2018-08-07:
I use linux but the issue is that it wont open at all
I dont know if this relates at all but i cant find the solution to this error
javaldx: Could not find a Java Runtime Environment!
Please ensure that a JVM and the package libreoffice-java-common
is installed.
If it is already installed then try removing ~/.libreoffice/3/user/config/javasettings_Linux_*.xml
Warning: failed to read path from javaldx
LibreOffice 6.0.3.2 00m0(Build:2)

Error in option: --Processing

Usage: soffice [argument...]
       argument - switches, switch parameters and document URIs (filenames).

Using without special arguments:
Opens the start center, if it is used without any arguments.
   {file}              Tries to open the file (files) in the components
                       suitable for them.
   {file} {macro:///Library.Module.MacroName}
                       Opens the file and runs specified macros from
                       the file.

Getting help and information:
   --help | -h | -?    Shows this help and quits.
   --helpwriter        Opens built-in or online Help on Writer.
   --helpcalc          Opens built-in or online Help on Calc.
   --helpdraw          Opens built-in or online Help on Draw.
   --helpimpress       Opens built-in or online Help on Impress.
   --helpbase          Opens built-in or online Help on Base.
   --helpbasic         Opens built-in or online Help on Basic scripting
                       language.
   --helpmath          Opens built-in or online Help on Math.
   --version           Shows the version and quits.
   --nstemporarydirectory
                       (MacOS X sandbox only) Returns path of the temporary
                       directory for the current user and exits. Overrides
                       all other arguments.

General arguments:
   --quickstart[=no]   Activates[Deactivates] the Quickstarter service.
   --nolockcheck       Disables check for remote instances using one
                       installation.
   --infilter={filter} Force an input filter type if possible. For example:
                       --infilter="Calc Office Open XML"
                       --infilter="Text (encoded):UTF8,LF,,,"
   --pidfile={file}    Store soffice.bin pid to {file}.
   --display {display} Sets the DISPLAY environment variable on UNIX-like
                       platforms to the value {display} (only supported by a
                       start script).

User/programmatic interface control:
   --nologo            Disables the splash screen at program start.
   --minimized         Starts minimized. The splash screen is not displayed.
   --nodefault         Starts without displaying anything except the splash
                       screen (do not display initial window).
   --invisible         Starts in invisible mode. Neither the start-up logo nor
                       the initial program window will be visible. Application
                       can be controlled, and documents and dialogs can be
                       controlled and opened via the API. Using the parameter,
                       the process can only be ended using the taskmanager
                       (Windows) or the kill command (UNIX-like systems). It
                       cannot be used in conjunction with --quickstart.
   --headless          Starts in "headless mode" which allows using the
                       application without GUI. This special mode can be used
                       when the application is controlled by external clients
                       via the API.
   --norestore         Disables restart and file recovery after a system crash.
   --safe-mode         Starts in a safe mode, i.e. starts temporarily with a
                       fresh user profile and helps to restore a broken
                       configuration.
[... 63 lines cut ...]
                   --convert-to pdf:writer_pdf_Export --outdir /home/user *.doc
                   --convert-to "html:XHTML Writer File:UTF8" *.doc
                   --convert-to "txt:Text (encoded):UTF8" *.doc
   --print-to-file [--printer-name printer_name] [--outdir output_dir]
                       Batch print files to file. If --outdir is not specified,
                       then current working directory is used as output_dir.
                       If --printer-name or --outdir used multiple times, only
                       last value of each is effective. Also, {Printername} of
                       --pt switch interferes with --printer-name.
   --cat               Dump text content of the following files to console
                       (implies --headless). Cannot be used with --convert-to.
   --script-cat        Dump text content of any scripts embedded in the files to console
                       (implies --headless). Cannot be used with --convert-to.
   -env:<VAR>[=<VALUE>] Set a bootstrap variable. For example: to set
                       a non-default user profile path:
                       -env:UserInstallation=file:///tmp/test

Ignored switches:
   -psn                Ignored (MacOS X only).
   -Embedding          Ignored (COM+ related; Windows only).
   --nofirststartwizard Does nothing, accepted only for backward compatibility.
   --protector {arg1} {arg2}
                       Used only in unit tests and should have two arguments.

any help pls i need it

closed 2019-04-20 by retiutut