openbci-archive/Docs #181: Clarify the Ganglion + WiFi Shield recommended power configuration

produceconsumerobot, 2019-02-26:
The recommended power configuration could be better explained to give users a "here's how to do what works" simple explanation.

retiutut, 2019-02-26:
- Power both separately
- Power up Ganglion
- Power up Wi-Fi shield
✅

produceconsumerobot, 2019-02-26:
@retiutut I think that's pretty clear in the general overview, but then when we get into the `Powering the Shield` section I think the combination of text and images could be clearer. For example, there's no unambiguous image with 2 battery packs clearly plugged in. The reference to "two batteries" also corresponds to an image that has 2 AA batteries shown, which could be a tad confusion because we actually mean "two battery packs" in that sentence. Additionally, I think the last pic/text about "on some modules this might work" might just add confusion for some users and perhaps should be removed entirely.

retiutut, 2019-05-07:
Addressed by #200, might still review this based on the information provided above

retiutut, 2019-05-12:
Updated Steps:

In general the steps are:

Power off WiFi Shield and Ganglion
Seat WiFi Shield on Ganglion
Put EXT PWR switch on Shield to OFF
Plug battery into Ganglion
Plug battery into WiFi Shield
Main power switch on Ganglion to ON
Main power switch on WiFi Shield to ON

retiutut, 2019-05-12:
I'm going to leave the instructions the way they were. The process describes seems to work.

closed 2019-05-12 by retiutut