openbci-archive/Docs #183: Simblee Quick Start Guide Link Broken

rptallman, 2019-02-26:
Hello, 

I'm trying to follow the [Ganglion Programming Tutorial](http://docs.openbci.com/Hardware/09-Ganglion_Programming_Tutorial#ganglion-programming-tutorial-setting-up-to-program-ganglion-using-hardware) but I am stuck at installing the Simblee software. Under the "What You Need" section, the instructions say to "follow the instructions in the [Simblee Quick Start Guide](https://www.simblee.com/Simblee_Quickstart_Guide_v1.1.0.pdf) to install the Simblee board files in Arduino for macOS" but that link is not working for me. I tried opening the link in Chrome and Safari but the website brings me to an error page. 

I believe this software is essential to following the rest of the tutorial...
Can someone help me get a new link to the Simblee OTA software? I can't find it myself.

This is my first time posting an issue and I hope I followed the 'rules'.

Thanks in advance

shirleyzhang867, 2019-02-27:
Hello, 
here's a [link](https://drive.google.com/file/d/1aZijlZ081pie5su9li1vdSes1nh_IMtv/view?usp=sharing) to the zipped folder containing the Simblee board files.

rptallman, 2019-03-12:
Hey, thanks for the link. I moved the simblee folder into the libraries folder in Arduino.

I'm running into another error. [This video](https://www.youtube.com/watch?time_continue=53&v=DHJm_49iL5Y) and [this video](https://www.youtube.com/watch?v=_8l2ziZgdps) are two quick start guides on youtube that I tried to follow, but I get this error when trying to add the simblee board in the Arduino settings:
[image]

My goal of all this is to get an LED to blink based on EEG control. [This tutorial](http://docs.openbci.com/Tutorials/07-Digital_Output_8bit_Example) has code to just make the LEDs blink with the Ganglion board and that is the first goal of mine right now. 

I appreciate the help if you have some time. I am struggling here :(

commit 1ed43792 (2019-05-12): Fix #183
commit 5ca50737 (2019-05-12): Merge pull request #204 from retiutut/master

Fix #183
closed 2019-05-12 by retiutut