openbci-archive/Docs #188: Cyton tutorial: bad formatting on section 6, "Optional Settings"

wjcroft, 2019-03-30:
Recently this section was changed to be hidden by default, but revealed with a dropdown arrow.

After clicking the arrow, the text formatting is hard to read because it sometimes overlaps with the pictures. Or is formatted oddly.

This is apparently a glitch in the Markdown to static page conversion software. Because the page looks fine when viewed from Github,

https://github.com/OpenBCI/Docs/blob/master/Tutorials/01-Cyton_Getting%20Started_Guide.md

wjcroft, 2019-03-30:
Additionally, the formatting bug seems to be related to the browser window width, try varying the width back and forth with your mouse, and you'll see what I'm referring to.

produceconsumerobot, 2019-03-31:
Thanks @wjcroft. This is a known issue pending upgrade to a new markdown interpreter that's currently under investigation.

wjcroft, 2019-03-31:
Sean, thanks. I only mentioned it because a forum member needed to change their channel. And in giving them the link to section 6, I noticed it was wonky. Before the drop down arrow was added to this section it it was fine. So a temp workaround might be to remove the dropdown feature. Regards,

retiutut, 2019-07-15:
Solution: Change markdown interpreter for the Docs website. The dropdowns are working on GitHub.

retiutut, 2019-08-08:
New docs repo has been created with updated markdown interpreter.

closed 2019-08-08 by retiutut