openbci-archive/Docs #194: Receiving

Smith-cj, 2019-04-18:


Smith-cj, 2019-04-18:
I'm using the command inlet.pull_sample() in Matlab on a marker inlet stream. The paradigm I'm using has two marker types, 1 and 2. The paradigm is in E-prime, and randomly picks between the two trial types. However, a marker is only received when the trial type changes. That is, if there are three trials of type 1 in a row, only the first of those trials is recorded, and no markers are received until a trial of type 2 starts, and so on. We checked with our BrainVision software, and the markers are indeed there, but aren't being received in Matlab.  Is there an easy work around with the inlet.pull_sample() function?

retiutut, 2019-04-18:
Not sure what this has to do with the Docs. Closing this issue.

closed 2019-04-18 by retiutut