openbci-archive/Docs #211: Consider separating out Win/Mac/Linux/Processing in OpenBCI_GUI instructions

produceconsumerobot, 2019-05-16:
It might add clarity to the instructions if I could look only at instructions for Windows without the need to be skipping over Mac/Linux/Processing instructions and risking missing a critical step in the process.

retiutut, 2019-05-16:
One side effect might be duplication of steps. Maybe the drop downs can be used more.

retiutut, 2019-05-22:
I think this is the way to go! I'm tired of looking for windows/mac/linux sections when updating the docs.

Proposed Solution: OpenBCI GUI doc with Links to 3 docs, one for each OS

retiutut, 2019-06-14:
Revised Solution: Separate OpenBCI GUI doc into **walkthrough and troublshooting**

retiutut, 2019-08-01:
Additional Requirement: Networking How-To and data output list

closed 2019-08-08 by retiutut