openbci-archive/Docs #237: Messed Ultracortex MarkIV Large Quads Naming

VIVelev, 2019-09-08:
The names of the large quads in https://github.com/OpenBCI/Docs/tree/master/assets/MarkIV/STL_Directory/Mark_IV_Frame/Mark_IV_Frame_Quads are messed up.

*M4_Large_Back_Left* **should be** *M4_Large_Front_Right*
*M4_Large_Back_Right* **should be** *M4_Large_Back_Left*
*M4_Large_Front_Left* **should be** *M4_Large_Back_Right*
*M4_Large_Front_Right* **should be** *M4_Large_Front_Left*

VIVelev, 2019-09-08:
Do you want me to pull request with the fix?

retiutut, 2019-09-08:
These Docs are no longer active.

retiutut, 2019-09-08:
https://github.com/OpenBCI/Documentation

retiutut, 2019-09-08:
Where did you find a link to these files?

VIVelev, 2019-09-08:
I found them here: https://openbci.github.io/Documentation/docs/04AddOns/01-Headwear/MarkIV#3d-printed-parts-1

> You can also print the frame in quarters or print the full frame. The .stl files are found here.

I needed these files because my 3D printer was too small to print the MarkIV by halves.

VIVelev, 2019-09-08:
I believe it will be usefull to add them to the new Docs also :)

retiutut, 2019-09-08:
@VIVelev You're welcome to make the Pull Request with this info. Thank you for noticing this.

VIVelev, 2019-09-08:
Great, I'm going to pr as soon as possible.

Maybe add the files to `Documentation/docs/assets/MarkIV` ?

VIVelev, 2019-09-08:
@retiutut Wouldn’t it be better to add the files to the [Ultracortex](https://github.com/OpenBCI/Ultracortex) repo under the folder `MarkIV`, since it's a more relevant place for those type of files.

retiutut, 2019-09-08:
@VIVelev So I guess the file Ultracortex Mark IV file names need to be changed there then? And then we change the links in the https://github.com/OpenBCI/Documentation  to the https://github.com/OpenBCI/Ultracortex repo.

VIVelev, 2019-09-08:
Yes, pretty much.

The STL files for the quads need to be added to `/Ultracortex/Mark_IV/MarkIV-FINAL/STL_Directory/` named properly. And then, as you said, we change the links in the https://github.com/OpenBCI/Documentation to the https://github.com/OpenBCI/Ultracortex repo.

retiutut, 2019-09-09:
@VIVelev Confirmed this naming bug just now. Looks like just for Large size yeah?

VIVelev, 2019-09-09:
@retiutut Also for the Medium, Small sizes: Left and Right should be swapped respectively.
(Front_Left -> Front_Right, Front_Right -> Front_Left and the same for Back)

VIVelev, 2019-09-09:
@retiutut I am going to PR the right naming in a minute or so to the Ultracortex repo

retiutut, 2019-09-09:
@VIVelev All you have to do is "fork" the repo, make changes locally, push changes to your branch, then make a Pull Request. I can help with changing the links in the new Docs.

retiutut, 2019-09-09:
@VIVelev No rush it's still the weekend lol

VIVelev, 2019-09-09:
@retiutut Yeah, no worries.

I am high school student and I am in a summer vacation still. 
I have plenty of time. :D

closed 2022-02-11 by VIVelev