openbci-archive/Docs #238: Board cover STL file has non-manifold edges and self-intersecting surfaces.

arge-7, 2021-02-05:
These errors cause problems with Simplify3D's slicing and I can't print the file without some of the edges getting messed up.

ThreeForm, 2021-02-06:
Simplify3D is one of the slicers used in normal production of Ultracortex parts, so the file has been verified to work. I also just double-checked in Netfabb. The file may have been corrupted during download. I suggest you[ download the full DevKit ](https://www.dropbox.com/s/7g9to9lef9cjcmc/OpenBCI%20Ultracortex%20M4_Dev%20Kit_ver0.5.zip?dl=0)to ensure you have everything.

arge-7, 2021-02-06:
[image]

Thanks for the quick reply. I'll try the DevKit files you uploaded. Here are some of the issues in the file downloaded from Github.

ThreeForm, 2021-02-06:
I zoomed in on those corners and there are very skinny triangles there. I
have attached a welded version from Netfabb, but I'm curious if you tried
printing it or if you stopped at the warning. You're the first person in
five years to report this issue so I'm interested in getting to the bottom
of it.
  -Aaron

arge-7, 2021-02-06:
Ha, that's funny to hear. I actually only noticed it when it was printing. I don't routinely check for errors like that, so I just hit print with my usual settings that have worked great for every other ultra cortex part I've printed so far. Then when I checked on the print shortly after starting, I noticed it was trying to print into thin air like the print had failed. That prompted me to compare the 3D model with the print preview code, and I noticed that these two areas were just disappearing from the gcode. Thanks again for being so responsive.

[image]

[image]

ThreeForm, 2021-02-06:
I gave it a closer look, and it turns out the files are different. The
screenshot below from Meshlab is from the STL I just downloaded from
GitHub, but it was from the Dev Kit.
[image: LightCase_19H_Cover00.png]
Notice how the one in your screenshot continues the step, whereas the one
above keeps the constant wall thickness. It looks like someone messed with
the file, which is something I can't really prevent when I hand the file
off to OpenBCI. They also strip out the version number from the filenames
so there's no way to tell what version anything is, so maybe this is just
an outdated file. I don't have publishing ability on Github but I'll let
someone know to replace the cover with  the one from the Dev Kit. Thanks
for your attention to detail in helping me figure this out.
 -Aaron

arge-7, 2021-02-06:
> I gave it a closer look, and it turns out the files are different. The screenshot below from Meshlab is from the STL I just downloaded from GitHub, but it was from the Dev Kit. [image: [...]

Weird, well at least you figured it out. Thanks again for helping me figure it out so quickly. I'm not seeing that welded netfabb file you mentioned attaching above, can you try sharing it again somehow? Have a great weekend.

closed 2022-09-29 by retiutut